yoctopuce/yoctolib_cpp #18: Compilation ARMEL demo

tim2000, 2016-08-09:
There is something I havn't understood.

Every GNUmakefile use the regular g++ to compile arm version of the demos. 

Since the lib GNUmakefile is ok, I don't understand what have been done. 
I've edited a lot of the GNUmakefile to make it compile for armel, it was a lot of pain.

yoctopuce, 2016-08-09:
Normaly you only need to run `make armel` in the `Examples/Doc-GettingStarted-Yocto-Buzzer` directory to compile the example. 

What modification have you done?

tim2000, 2016-08-09:
First, I use my compiler (arm-linux-gnueabi-gcc), not excatly the same as you, so I've forced it.

Then I've read this in the makefile :
```
$(DEMO_ARMEL) : main.cpp $(YOCTO_API_DIR_ARMEL)*  $(DIR_ARMEL)
	@g++ $(OPTS_GENERIC) $(OPTS_ARMEL) -o $@ main.cpp -L$(YOCTO_API_DIR_ARMEL) $(OPTS_LINK)
```
It's ARMEL compile line, using classic g++, so I've done :

```
CC_ARMEL = <path to my compiler>/arm-linux-gnueabi-gcc
CPP_ARMEL = <path to my compiler>/arm-linux-gnueabi-g++

$(DEMO_ARMEL) : main.cpp $(YOCTO_API_DIR_ARMEL)*  $(DIR_ARMEL)
	@$(CPP_ARMEL) $(OPTS_GENERIC) $(OPTS_ARMEL) -o $@ main.cpp -L$(YOCTO_API_DIR_ARMEL) $(OPTS_LINK)
```

I've also remove -mfloat-abi, -march and -marm since these options doesn't exist with my gcc.

yoctopuce, 2016-08-10:
We have written the Makefile in such manner that you compile the example directly on the host that will run the binary. Ex you download, compile and run the example directly on the Raspberry PI.

So the Makefiles of the examples are intentionally been kept simple in order to be understandable by someone that is not a GNU Makefile expert.  The idea was that people that want to do cross compilation have the skill to understand and tweak the makefile.

Moreover been able to detect if the host is armel or armhf is not trivial.

tim2000, 2016-08-10:
Ok, this is just not for cross-compilation then. I understand.

closed 2016-08-10 by tim2000