yoctopuce/yoctolib_cpp #20: Deadlock with two USB PWM Rx devices during extended insertion/removal tests

rk-bp, 2017-01-16:
unit_test-Multitech Debug [C/C++ Remote Application] 
unit_test [5558] [cores: 0] (0 filtered running threads) 
Thread #1 5558 [core: 0] (Suspended : Signal : SIGINT:Interrupt) 
__lll_lock_wait() at lowlevellock.c:46 0xb6fc6a9c 
__GI___pthread_mutex_lock() at pthread_mutex_lock.c:114 0xb6fc0ec8 
Poco::MutexImpl::lockImpl() at Mutex_POSIX.h:61 0x122fc 
Poco::Mutex::lock() at Mutex.h:209 0x124bc 
Poco::ScopedLock<Poco::Mutex>::ScopedLock() at ScopedLock.h:40 0x130c4 
PollingThread::Stop() at PollingThread.cpp:27 0x37448 
DevicePort::Stop() at DevicePort.h:18 0x350e4 
PulsePort::Stop() at PulsePort.cpp:175 0x43338 
PulseDevice::Stop() at PulseDevice.cpp:100 0x39ee4 
PulseTests_FunctionalTest_Test::TestBody() at PulseTests.cpp:461 0x1cd60 
<...more frames...> 

Thread #2 6272 [core: 0] (Suspended : Container) 
__pthread_cond_wait() at pthread_cond_wait.c:187 0xb6fc383c 
Poco::EventImpl::waitImpl() at Event_POSIX.cpp:99 0x6e328 
wait() at Event.h:99 0x72f64 
Poco::PooledThread::run() at ThreadPool.cpp:193 0x72f64 
Poco::(anonymous namespace)::RunnableHolder::run at Thread.cpp:57 0x66354 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x66488 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6af6878 
() at clone.S:92 0xb6af6878 

Thread #3 6273 [core: 0] (Suspended : Container) 
__pthread_cond_wait() at pthread_cond_wait.c:187 0xb6fc383c 
Poco::EventImpl::waitImpl() at Event_POSIX.cpp:99 0x6e328 
wait() at Event.h:99 0x72f64 
Poco::PooledThread::run() at ThreadPool.cpp:193 0x72f64 
Poco::(anonymous namespace)::RunnableHolder::run at Thread.cpp:57 0x66354 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x66488 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6af6878 
() at clone.S:92 0xb6af6878 

Thread #4 9997 [core: 0] (Suspended : Container) 
poll() at syscall-template.S:81 0xb6ae85d0 
handle_events() at io.c:1,888 0xb6d15e38 
libusb_handle_events_timeout_completed() at io.c:2,024 0xb6d167e8 
event_thread() at ypkt_lin.c:254 0xf10b8 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6af6878 
() at clone.S:92 0xb6af6878 

Thread #5 9998 [core: 0] (Suspended : Container) 
__pthread_cond_destroy() at pthread_cond_destroy.c:75 0xb6fc343c 
yCloseEvent() at ythread.c:223 0x905e8 
yPktQueueFree() at ystream.c:935 0xd60a4 
yyyPacketShutdown() at ypkt_lin.c:665 0xf2204 
yyPacketSetup() at ystream.c:1,367 0xd719c 
yPacketSetup() at ystream.c:1,425 0xd742c 
yStreamSetup() at ystream.c:1,538 0xd792c 
StartDevice() at ystream.c:2,091 0xd92c8 
enuUpdateDStatus() at ystream.c:2,175 0xd9640 
yUSBUpdateDeviceList() at ystream.c:2,320 0xda004 
yapiUpdateDeviceList_internal() at yapi.c:2,597 0xcf3e8 
yapiUpdateDeviceList() at yapi.c:4,385 0xd3d48 
YapiWrapper::updateDeviceList() at yocto_api.cpp:3,739 0x9fc20 
YDevice::requestAPI() at yocto_api.cpp:2,536 0x9c470 
YFunction::isOnline() at yocto_api.cpp:2,239 0x9ab80 
PulseDevice::readValue<double>() at PulseDevice.cpp:498 0x3dc78 
PulseDevice::UpdateItemValues() at PulseDevice.cpp:192 0x3b39c 
PulseDevice::Poll() at PulseDevice.cpp:161 0x3a910 
PulsePort::Poll() at PulsePort.cpp:232 0x43d64 
PollingThread::Thread() at PollingThread.cpp:68 0x3794c 
PollingThread::StaticPollingThread() at PollingThread.cpp:51 0x377e8 
Poco::(anonymous namespace)::CallableHolder::run at Thread.cpp:80 0x66368 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x66488 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6af6878 
() at clone.S:92 0xb6af6878 

Thread #6 9999 [core: 0] (Suspended : Container) 
__lll_lock_wait() at lowlevellock.c:46 0xb6fc6a9c 
__GI___pthread_mutex_lock() at pthread_mutex_lock.c:79 0xb6fc0d20 
yEnterCriticalSection() at ythread.c:601 0x90aac 
devPauseIO() at ystream.c:496 0xd5750 
yUsbReadBlock() at ystream.c:2,832 0xdb358 
yapiRequestWaitEndUSB() at yapi.c:3,238 0xd1524 
yapiHTTPRequestSyncStartEx_internal() at yapi.c:3,326 0xd18b8 
yapiHTTPRequestSyncStartOutOfBand() at yapi.c:4,549 0xd4298 
YDevice::HTTPRequest() at yocto_api.cpp:2,501 0x9c140 
YDevice::requestAPI() at yocto_api.cpp:2,533 0x9c450 
YFunction::isOnline() at yocto_api.cpp:2,239 0x9ab80 
PulseDevice::readValue<double>() at PulseDevice.cpp:498 0x3dc78 
PulseDevice::UpdateItemValues() at PulseDevice.cpp:192 0x3b39c 
PulseDevice::Poll() at PulseDevice.cpp:161 0x3a910 
PulsePort::Poll() at PulsePort.cpp:232 0x43d64 
PollingThread::Thread() at PollingThread.cpp:68 0x3794c 
PollingThread::StaticPollingThread() at PollingThread.cpp:51 0x377e8 
Poco::(anonymous namespace)::CallableHolder::run at Thread.cpp:80 0x66368 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x66488 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6af6878 
() at clone.S:92 0xb6af6878 
Remote Shell 
/opt/mlinux/3.2.3/sysroots/x86_64-mlinux-linux/usr/bin/arm-mlinux-linux-gnueabi/arm-mlinux-linux-gnueabi-gdb (7.6.2)

yoctopuce, 2017-01-17:
Can describe your issue? It's impossible to understand your problem without minimal informations on what your are doing. 

- Which OS are you using?
- How did you have compiled your application
- Which device are you using?
- What are doing your code?
- Does it append every time?
- Do you have tested on other computer?
- ....

rk-bp, 2017-01-30:
Hello,

Which OS are you using?

We are using Linux 3.12 on an ARM V5 based embedded system.  The version of libusb is v1.0.21.

How did you have compiled your application

We build a distro for the embedded system using Yocto (bitbake) for which we have generated a bitbake recipe for Yoctopuce C++ build based around the build.sh script included with Yoctopuce C++.  From that an SDK is generated and we cross-compile using Eclipse IDE to link in the Yoctopuce static libraries.

Which device are you using?
ARM V5

What are doing your code?

We have our ARM V5 embedded system plugged into a four-port USB hub and on that USB hub we have a Yocto-PWM-Tx and two Yocto-PWM-Rx.  One output of the Yocto-PWM-Tx is feeding the first PWM input on one Yocto-PWM-Rx and one output is feeding the first PWM input on the other Yocto-PWM-Rx.  The Yocto-PWM-Tx outputs are set to 100 Hz at 50% duty cycle.  There is a thread each spun up to read the pulses, frequency and uptime on each of the Yocto-PWM-Rx.  Before reading the values, a call is made to see if the module is online and then if the PWM input is online.  The main thread fires up the two Yocto-PWM-Rx threads after setting up the Yocto-PWM-Tx.  These threads read all of the aforementioned params and queue the results back to the main thread where they are displayed.  Every ten seconds, the main thread sets the reboot count to two seconds for each Yocto-PWM-Rx to force a reboot.  The test runs endlessly until user hits Ctrl-C.

What was uncovered were mutual exclusion issues, a short circuit return not releasing a mutex, a race condition freeing USB buffers before libusb_close() is called (segfault) and not resetting USB device to a known stated prior to opening it (see attached patch to your 1/13/2017 release).

Does it append every time?

I don't understand this question.

Do you have tested on other computer?

Yes.

rk-bp, 2017-01-30:
GitHub would not let me attach a .zip file of the patch.

rk-bp, 2017-01-30:
diff -Naur --strip-trailing-cr '--exclude=.git' '--exclude=*.pyc' '--exclude=*.py' '--exclude=*.*~' Sources/yapi/yapi.c Sources.new/yapi/yapi.c
--- Sources/yapi/yapi.c	2016-06-06 14:48:16.000000000 -0400
+++ Sources.new/yapi/yapi.c	2017-01-25 10:50:09.435372000 -0500
@@ -2634,6 +2634,18 @@
 u64 test_pkt=0;
 u64 test_tout=0;
 
+#ifndef NOMINMAX
+
+#ifndef max
+#define max(a,b)            (((a) > (b)) ? (a) : (b))
+#endif
+
+#ifndef min
+#define min(a,b)            (((a) < (b)) ? (a) : (b))
+#endif
+
+#endif  /* NOMINMAX */
+
 static YRETCODE  yapiSleep_internal(int ms_duration, char *errmsg)
 {
     u64     now, timeout;
@@ -2651,7 +2663,7 @@
         now =yapiGetTickCount();
         // todo: we may want to use a samller timeout
         if (now < timeout) {
-           if (yWaitForEvent(&yContext->exitSleepEvent, (int) (timeout - now)))
+           if (yWaitForEvent(&yContext->exitSleepEvent, max((int) (timeout - now), 0)))
                test_pkt++; //just for testing
            else
                test_tout++;
diff -Naur --strip-trailing-cr '--exclude=.git' '--exclude=*.pyc' '--exclude=*.py' '--exclude=*.*~' Sources/yapi/ypkt_lin.c Sources.new/yapi/ypkt_lin.c
--- Sources/yapi/ypkt_lin.c	2016-12-01 10:25:14.000000000 -0500
+++ Sources.new/yapi/ypkt_lin.c	2017-01-29 19:25:08.090868400 -0500
@@ -212,7 +212,10 @@
     res=libusb_control_transfer(hdl, LIBUSB_ENDPOINT_IN,
         LIBUSB_REQUEST_GET_DESCRIPTOR, (LIBUSB_DT_STRING << 8) | desc_index,
         0, buffer, 512, 10000);
-    if(res<0) return res;
+	if (res < 0) {
+		yLeaveCriticalSection(&yContext->string_cache_cs);
+		return res;
+	}
 
     len=(buffer[0]-2)/2;
     if (len >= length) {
@@ -515,6 +518,20 @@
     if(iface->devref==NULL){
         return YERR(YAPI_DEVICE_NOT_FOUND);
     }
+
+	// we need to do this as it is possible that the device was not closed properly in a previous session
+	// if we don't do this and the device wasn't closed properly odd behavior results.
+    if((res=libusb_open(iface->devref,&iface->hdl))!=0){
+        return yLinSetErr("libusb_open", res,errmsg);
+    }
+    else
+	{
+		libusb_reset_device(iface->hdl);
+		libusb_close(iface->hdl);
+
+		usleep(200);
+	}
+
     if((res=libusb_open(iface->devref,&iface->hdl))!=0){
         return yLinSetErr("libusb_open", res,errmsg);
     }
@@ -632,6 +649,7 @@
 {
     int res,j;
[... 222 lines cut ...]
         return YAPI_IO_ERROR;
     }
     // we know for sure that the last character parsed was a '{'
@@ -2590,6 +2622,7 @@
     // store result in cache
     _cacheJson = string(j.src);
     _cacheStamp = yapiGetTickCount() + YAPI::DefaultCacheValidity;
+	yLeaveCriticalSection(&_updateDeviceList_CS);
 
     return YAPI_SUCCESS;
 }
diff -Naur --strip-trailing-cr '--exclude=.git' '--exclude=*.pyc' '--exclude=*.py' '--exclude=*.*~' Sources/yocto_api.h Sources.new/yocto_api.h
--- Sources/yocto_api.h	2017-01-11 09:51:46.000000000 -0500
+++ Sources.new/yocto_api.h	2017-01-30 09:05:36.028093300 -0500
@@ -1586,9 +1586,9 @@
      * Returns the current value of a single function attribute, as a text string, as quickly as
      * possible but without using the cached value.
      *
-     * @param attrName : le nom de l'attribut désiré
+     * @param attrName : le nom de l'attribut d괩r銠     *
-     * @return une chaîne de caractères représentant la valeur actuelle de l'attribut.
+     * @return une chaﯥ de caract鳥s repr괥ntant la valeur actuelle de l'attribut.
      *
      * On failure, throws an exception or returns an empty string.
      */

rk-bp, 2017-01-30:
`diff -Naur --strip-trailing-cr '--exclude=.git' '--exclude=*.pyc' '--exclude=*.py' '--exclude=*.*~' Sources/yapi/yapi.c Sources.new/yapi/yapi.c
--- Sources/yapi/yapi.c	2016-06-06 14:48:16.000000000 -0400
+++ Sources.new/yapi/yapi.c	2017-01-25 10:50:09.435372000 -0500
@@ -2634,6 +2634,18 @@
 u64 test_pkt=0;
 u64 test_tout=0;
 
+#ifndef NOMINMAX
+
+#ifndef max
+#define max(a,b)            (((a) > (b)) ? (a) : (b))
+#endif
+
+#ifndef min
+#define min(a,b)            (((a) < (b)) ? (a) : (b))
+#endif
+
+#endif  /* NOMINMAX */
+
 static YRETCODE  yapiSleep_internal(int ms_duration, char *errmsg)
 {
     u64     now, timeout;
@@ -2651,7 +2663,7 @@
         now =yapiGetTickCount();
         // todo: we may want to use a samller timeout
         if (now < timeout) {
-           if (yWaitForEvent(&yContext->exitSleepEvent, (int) (timeout - now)))
+           if (yWaitForEvent(&yContext->exitSleepEvent, max((int) (timeout - now), 0)))
                test_pkt++; //just for testing
            else
                test_tout++;
diff -Naur --strip-trailing-cr '--exclude=.git' '--exclude=*.pyc' '--exclude=*.py' '--exclude=*.*~' Sources/yapi/ypkt_lin.c Sources.new/yapi/ypkt_lin.c
--- Sources/yapi/ypkt_lin.c	2016-12-01 10:25:14.000000000 -0500
+++ Sources.new/yapi/ypkt_lin.c	2017-01-29 19:25:08.090868400 -0500
@@ -212,7 +212,10 @@
     res=libusb_control_transfer(hdl, LIBUSB_ENDPOINT_IN,
         LIBUSB_REQUEST_GET_DESCRIPTOR, (LIBUSB_DT_STRING << 8) | desc_index,
         0, buffer, 512, 10000);
-    if(res<0) return res;
+	if (res < 0) {
+		yLeaveCriticalSection(&yContext->string_cache_cs);
+		return res;
+	}
 
     len=(buffer[0]-2)/2;
     if (len >= length) {
@@ -515,6 +518,20 @@
     if(iface->devref==NULL){
         return YERR(YAPI_DEVICE_NOT_FOUND);
     }
+
+	// we need to do this as it is possible that the device was not closed properly in a previous session
+	// if we don't do this and the device wasn't closed properly odd behavior results.
+    if((res=libusb_open(iface->devref,&iface->hdl))!=0){
+        return yLinSetErr("libusb_open", res,errmsg);
+    }
+    else
+	{
+		libusb_reset_device(iface->hdl);
+		libusb_close(iface->hdl);
+
+		usleep(200);
+	}
+
     if((res=libusb_open(iface->devref,&iface->hdl))!=0){
         return yLinSetErr("libusb_open", res,errmsg);
     }
@@ -632,6 +649,7 @@
 {
     int res,j;
[... 223 lines cut ...]
     }
     // we know for sure that the last character parsed was a '{'
@@ -2590,6 +2622,7 @@
     // store result in cache
     _cacheJson = string(j.src);
     _cacheStamp = yapiGetTickCount() + YAPI::DefaultCacheValidity;
+	yLeaveCriticalSection(&_updateDeviceList_CS);
 
     return YAPI_SUCCESS;
 }
diff -Naur --strip-trailing-cr '--exclude=.git' '--exclude=*.pyc' '--exclude=*.py' '--exclude=*.*~' Sources/yocto_api.h Sources.new/yocto_api.h
--- Sources/yocto_api.h	2017-01-11 09:51:46.000000000 -0500
+++ Sources.new/yocto_api.h	2017-01-30 09:05:36.028093300 -0500
@@ -1586,9 +1586,9 @@
      * Returns the current value of a single function attribute, as a text string, as quickly as
      * possible but without using the cached value.
      *
-     * @param attrName : le nom de l'attribut désiré
+     * @param attrName : le nom de l'attribut d괩r銠     *
-     * @return une chaîne de caractères représentant la valeur actuelle de l'attribut.
+     * @return une chaﯥ de caract鳥s repr괥ntant la valeur actuelle de l'attribut.
      *
      * On failure, throws an exception or returns an empty string.
      */
`

rk-bp, 2017-01-30:
[Sources.patch.txt](https://github.com/yoctopuce/yoctolib_cpp/files/739232/Sources.patch.txt)

yoctopuce, 2017-02-02:
Our library is currently not designed to be used by multiples thread in the same time. Having multiples
thread that use the USB port can lead to some performances issue or bugs. We know that it's a 
big limitation and we are working on a fix.

However in your case, the root cause of your dead lock seems related to a device that is not starting
correctly. In this case we normally log the cause of the issue. Could you register a 
debug log function with the `YAPI::RegisterLogFunction()` and post the debug messages?

Regarding your patch file, your change in the yyySetup function is very interesting. We never
needed to reset the interface, so it's seems that there something strange with your USB port.
Is you USB hub powered? By experience we know that most of the USB bug that customers have
are related to a non powered USB hub or a too long cable. We have written two articles on this
subject:
 * http://www.yoctopuce.com/EN/article/how-many-usb-devices-can-you-connect
 * http://www.yoctopuce.com/EN/article/connecting-your-yoctopuce-modules

Also is it possible to get your source code? I'm trying to reproduce your crash on a 
Raspberry Pi and I may speed up thing if we use the same code. If you do not want 
to share your code on github you can contact support@yoctopuce.com

rk-bp, 2017-02-02:
When I looked at your source initially, I was under the impression that it could be used in a multi-threaded environment as several functions utilized mutex protection.  I quickly found out that additional protection was necessary.

I am utilizing a USB hub (I have tried hubs (USB 3.0) from two manufacturers) and two PWM Rx and one PWM Tx.  My test is running (with patch I submitted a few days ago) each PWM Rx on it's own thread being sourced signals from a single PWM Tx on the main thread.  Each is constantly being read on it's own thread and rebooted a few seconds into each test cycle.  I noticed that both would become unresponsive a couple of hours after initiating the test without resetting them in the yyySetup function.  Yes, I am using non-powered USB hubs.  However, since libusb essentially allows implementation of a USB device driver in user space, I believe that it is perfectly acceptable to reset the device on setup so that it is in a "known" initial state.

I also noticed that the transfer buffers need to be freed after the call to libusb_close() to avoid a segfault as the transfer buffers still seemed to be in use (in flight) by libusb even though an attempt is made to cancel the transfers prior to calling libusb_close().

I will see what I can do about getting you some code to reproduce these issues.  Right now, I am testing our source code with your libraries linked in Google Test.  I will try to make some time to come up with a simple example that demonstrates the issues noted in this bug report.

yoctopuce, 2017-03-20:
We have released a new build (v26849) that should be thread safe.  We have not used the same approach as you but the result should be the same. Let me know if you have any issue with this version.

rk-bp, 2017-03-22:
Unfortunately, this new version results in a segfault.

rk-bp, 2017-03-22:
unit_test-Multitech Debug [C/C++ Remote Application] 
unit_test [12667] [cores: 0] (0 filtered running threads) 
Thread #1 12667 [core: 0] (Suspended : Container) 
__lll_lock_wait() at lowlevellock.c:46 0xb6fc6a9c 
__GI___pthread_mutex_lock() at pthread_mutex_lock.c:114 0xb6fc0ec8 
Poco::MutexImpl::lockImpl() at Mutex_POSIX.h:61 0x1f2f8 
Poco::Mutex::lock() at Mutex.h:209 0x228d4 
Poco::ScopedLock<Poco::Mutex>::ScopedLock() at ScopedLock.h:40 0x2310c 
PulsePort::setSavePulseCount() at PulsePort.cpp:318 0x61960 
DevicePort::SavePulseCount() at DevicePort.h:32 0x2e81c 
PulseTests_FunctionalTest_Test::TestBody() at PulseTests.cpp:528 0x2cb08 
HandleSehExceptionsInMethodIfSupported<testing::Test, void>() at gtest.cc:2,078 0x513e24 
testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>() at gtest.cc:2,114 0x513e24 
<...more frames...> 
Thread #3 13510 [core: 0] (Suspended : Container) 
__pthread_cond_wait() at pthread_cond_wait.c:187 0xb6fc383c 
Poco::EventImpl::waitImpl() at Event_POSIX.cpp:108 0x41cfb0 
wait() at Event.h:99 0x437688 
Poco::PooledThread::run() at ThreadPool.cpp:193 0x437688 
Poco::(anonymous namespace)::RunnableHolder::run at Thread.cpp:57 0x4141a4 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x4142d8 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6aee878 
() at clone.S:92 0xb6aee878 
Thread #4 13511 [core: 0] (Suspended : Container) 
__pthread_cond_wait() at pthread_cond_wait.c:187 0xb6fc383c 
Poco::EventImpl::waitImpl() at Event_POSIX.cpp:108 0x41cfb0 
wait() at Event.h:99 0x437688 
Poco::PooledThread::run() at ThreadPool.cpp:193 0x437688 
Poco::(anonymous namespace)::RunnableHolder::run at Thread.cpp:57 0x4141a4 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x4142d8 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6aee878 
() at clone.S:92 0xb6aee878 
Thread #5 13574 [core: 0] (Suspended : Container) 
__pthread_cond_timedwait() at pthread_cond_timedwait.c:194 0xb6fc3c64 
Poco::EventImpl::waitImpl() at Event_POSIX.cpp:164 0x41d1ac 
tryWait() at Event.h:112 0x406738 
Poco::Timer::run() at Timer.cpp:187 0x406738 
Poco::PooledThread::run() at ThreadPool.cpp:200 0x437600 
Poco::(anonymous namespace)::RunnableHolder::run at Thread.cpp:57 0x4141a4 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x4142d8 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6aee878 
() at clone.S:92 0xb6aee878 
Thread #6 15625 [core: 0] (Suspended : Signal : SIGSEGV:Segmentation fault) 
__GI___pthread_mutex_lock() at pthread_mutex_lock.c:66 0xb6fc0c60 
yEnterCriticalSection() at ythread.c:648 0x1b0790 
yPktQueuePop() at ystream.c:1,030 0x18cf80 
yPktQueueWaitAndPopD2H() at ystream.c:1,129 0x18d334 
yGetNextPktEx() at ystream.c:1,405 0x18ded4 
yStreamReceived() at ystream.c:1,485 0x18e1ec 
yDispatchReceive() at ystream.c:1,864 0x18f5a8 
yUsbReadBlock() at ystream.c:2,736 0x19187c 
yapiRequestWaitEndUSB() at yapi.c:3,281 0x188120 
yapiHTTPRequestSyncStartEx_internal() at yapi.c:3,369 0x1884b4 
yapiHTTPRequestSyncStartOutOfBand() at yapi.c:4,592 0x18ae98 
YDevice::HTTPRequest_unsafe() at yocto_api.cpp:2,596 0x15159c 
YDevice::requestAPI() at yocto_api.cpp:2,657 0x151b18 
YFunction::isOnline() at yocto_api.cpp:2,334 0x150078 
PulseDevice::readValue<double>() at PulseDevice.cpp:552 0x58d8c 
PulseDevice::UpdateItemValues() at PulseDevice.cpp:266 0x564a0 
PulseDevice::Poll() at PulseDevice.cpp:235 0x55a64 
PulsePort::Poll() at PulsePort.cpp:237 0x60554 
PollingThread::Thread() at PollingThread.cpp:68 0x51fa4 
PollingThread::StaticPollingThread() at PollingThread.cpp:51 0x51e40 
Poco::(anonymous namespace)::CallableHolder::run at Thread.cpp:80 0x4141b8 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x4142d8 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6aee878 
[... 20 lines cut ...]
yyySetup() at ypkt_lin.c:519 0x1b1a1c 
yPacketSetup() at ystream.c:1,346 0x18dbcc 
yStreamSetup() at ystream.c:1,459 0x18e0f4 
StartDevice() at ystream.c:2,011 0x18faa8 
enuUpdateDStatus() at ystream.c:2,095 0x18fe20 
yUSBUpdateDeviceList() at ystream.c:2,234 0x190588 
yapiUpdateDeviceList_internal() at yapi.c:2,639 0x185f8c 
yapiUpdateDeviceList() at yapi.c:4,428 0x18a948 
YapiWrapper::updateDeviceList() at yocto_api.cpp:3,880 0x1553e4 
YFunction::_getDescriptor() at yocto_api.cpp:1,722 0x14c99c 
YFunction::_getDevice() at yocto_api.cpp:1,743 0x14ca4c 
YFunction::isOnline() at yocto_api.cpp:2,328 0x150030 
PulseDevice::readValue<double>() at PulseDevice.cpp:586 0x59254 
PulseDevice::UpdateItemValues() at PulseDevice.cpp:266 0x564a0 
PulseDevice::Poll() at PulseDevice.cpp:235 0x55a64 
PulsePort::Poll() at PulsePort.cpp:237 0x60554 
PollingThread::Thread() at PollingThread.cpp:68 0x51fa4 
PollingThread::StaticPollingThread() at PollingThread.cpp:51 0x51e40 
Poco::(anonymous namespace)::CallableHolder::run at Thread.cpp:80 0x4141b8 
Poco::ThreadImpl::runnableEntry() at Thread_POSIX.cpp:349 0x4142d8 
start_thread() at pthread_create.c:314 0xb6fbe424 
() at clone.S:92 0xb6aee878 
() at clone.S:92 0xb6aee878 
Remote Shell 
/opt/mlinux/3.2.3/sysroots/x86_64-mlinux-linux/usr/bin/arm-mlinux-linux-gnueabi/arm-mlinux-linux-gnueabi-gdb (7.6.2)

rk-bp, 2017-03-23:
It looks as though yPrivDeviceSt structure members can be manipulated on multiple threads.  This seems to be what happened in the trace I previously posted.  Two PWM Rx devices are being polled on separate threads and then randomly reset via your API.  I don't know criteria governs the lifespan of the device objects or if these structures (apparently in a linked list) persist through device plug/unplug and reset events.  If they do, then one could certainly invoke protection around the access to the structure via the acces_state critical section.  These are just my observations from briefly perusing the code.  :)

yoctopuce, 2017-03-31:
This bug has been fixed in the last release. Thanks for your help.

closed 2017-03-31 by yoctopuce