yoctopuce/yoctolib_cpp #21: YPwmInput::set_pulseCounter() does not work.

rk-bp, 2017-02-01:
YPwmInput::set_pulseCounter(10000) does not work in version of released on 1/13/2017.

"GET /api/pwmInput2/pulseCounter?pulseCounter=100000&. \r\n\r\n"

yoctopuce, 2017-02-02:
It's not as issue of the C++ library but a limitation in the firmware of the Yocto-PWM-RX.
Originally, the set_pulseCounter function was only design to reset the counter, so the function
discarded value different of zero.

 We have fixed this bug in the firmware v26535 which work now as expected.

 The firmware is available on our web site:
	http://www.yoctopuce.com/EN/products/usb-electrical-interfaces/yocto-pwm-rx

rk-bp, 2017-02-02:
Thanks for the quick fix.  I upgraded to the latest firmware and the set_pulseCounter function can now set values other than zero.

closed 2017-02-03 by yoctopuce