yoctopuce/yoctolib_cpp #22: Handle Leak in Linux CPP Source

rk-bp, 2017-02-03:
Extended testing revealed a handle leak in the 1/13/2017 version of Linux CPP source.  I fixed it below...well, the handle leak is not occurring anymore anyway.

#define YOCTO_LOCK_PIPE "/tmp/.yoctolock"

// return 1 if we can reserve access to the device 0 if the device
// is already reserved
static int yReserveGlobalAccess(yContextSt *ctx, char *errmsg)
{
    int fd;
    int chk_val, mypid, usedpid = 0;
    size_t res;
    mode_t mode=0666;
    mode_t oldmode = umask(0000);
    char msg[YOCTO_ERRMSG_LEN];

    HALLOG("old mode (%#o)\n",oldmode);
    HALLOG("create fifo with (%#o)\n",mode);
    if(mkfifo(YOCTO_LOCK_PIPE,mode)<0) {
        HALLOG("unable to create lock fifo (%d:%s)\n",errno,strerror(errno));
    }
    umask(oldmode);
    fd = open(YOCTO_LOCK_PIPE,O_RDWR|O_NONBLOCK);
    if(fd<0){
        HALLOG("unable to open lock fifo (%d)\n",errno);
        if(errno==EACCES) {
            return YERRMSG(YAPI_DOUBLE_ACCES, "we do not have acces to lock fifo");
        }else{
            // we cannot open lock file so we cannot realy
            // check double instance so we asume that we are
            // alone
            return YAPI_SUCCESS;
        }
    }
    chk_val = 0;
    mypid = (int) getpid();
    res = read(fd, &chk_val, sizeof(chk_val));
    if (res == sizeof(chk_val)) {
        //there is allready someone
        usedpid = chk_val;
    } else{
        // nobody there -> store my PID
        chk_val = mypid;
    }
    res = write(fd, &chk_val, sizeof(chk_val));
    if(res != sizeof(chk_val)) {
        YSPRINTF(msg, YOCTO_ERRMSG_LEN, "Write to lock fifo failed (%d)", res);
		close(fd);
        return YERRMSG(YAPI_DOUBLE_ACCES, msg);
    }
    if (usedpid != 0) {
        if (usedpid == 1) {
			close(fd);
            // locked by api that not store the pid
            return YERRMSG(YAPI_DOUBLE_ACCES, "Another process is already using yAPI");
        } else {
            YSPRINTF(msg, YOCTO_ERRMSG_LEN, "Another process (pid %d) is already using yAPI", (u32) usedpid);
			close(fd);
            return YERRMSG(YAPI_DOUBLE_ACCES, msg);
        }
    }
	close(fd);
    return YAPI_SUCCESS;
}

size_t dropwarning;

static void yReleaseGlobalAccess(yContextSt *ctx)
{
    int chk_val;
    int fd = open(YOCTO_LOCK_PIPE,O_RDWR|O_NONBLOCK);
    if(fd>=0){
        dropwarning = read(fd,&chk_val,sizeof(chk_val));
		close(fd);
      }
}

rk-bp, 2017-02-06:
Cumulative patch for Yoctopuce issues with handle fix which persists the handle in the API context (for Linux only).

[yoctopuce.patch.txt](https://github.com/yoctopuce/yoctolib_cpp/files/755196/yoctopuce.patch.txt)

yoctopuce, 2017-02-21:
We have fixed this issue in the latest version of the lib v1.10.26625.

closed 2017-02-21 by yoctopuce