yoctopuce/yoctolib_cpp #24: For Yocto-3D-V2 it would be nice to have a way to get all sensor results together in a synchronized way

boilerbots, 2017-04-25:
To really use the IMU in an accurate and precise way you need to have some way to guarantee that all the sensor readings are synchronized across the device, for example 3 axis accel and 3 axis gyro.

Currently I don't see a way to do this. Can the firmware in the controller make sure that it keeps synchronized results and the software API provide a way to read these results.

yoctopuce, 2017-04-26:
Internally, the sensor always updates all values synchronously, and the API fetch them all at once as well (they are then cached for 5ms). So this should not be a problem.
This being said, don't expect "accurate and precise" values from such a simple MEMS sensor: it gives a good estimate of the device orientation, but this is not a navigation instrument :-)
In particular, don't expect to integrate acceleration to derive speed and then position: that would just not work, because the error would be integrated and lead to meaningless values. See https://www.yoctopuce.com/EN/article/accelerometers-and-gyroscopes-fantasy-vs-reality

boilerbots, 2017-04-26:
That is good to know. Digging a little deeper, I thought the IMU ran at 100Hz, but you say the values are cached for 5ms. Can you explain further? Is there a timestamp such that I could tell when the samples are refreshed?

The BNO055 is a very capable device, at least according to the datasheet, I wonder if we could get more out of it with some synchronous callback interface. Also looking for a way to disable the Magenetometers for orientation in the fusion algorithm, will create a different ticket if I need this.

yoctopuce, 2017-04-27:
You can find advices on how to get high-frequency readings in this article: https://www.yoctopuce.com/EN/article/tips-for-using-the-yocto-3d-v2

Regarding disabling the magnetometer in the fusion algorithm, this is not possible. You can ignore the compass if you don't care about orientation, but it is as hopeless to expect an angle estimate solely based on a simple MEMS gyroscope as it is hopeless to expect a speed estimate solely based on a simple MEMS accelerometer. Over the time, integrated errors make the value useless. Gyroscopes and accelerometers are excellent for reactive estimates, but for long-term estimates and stabilization you need an absolute reference.

boilerbots, 2017-05-02:
Thanks for the links, useful information that I didn't find in the document.

About disabling the magnetometer in the fusion algorithm. I am asking if we could select the fusion mode as described in the BNO055 document section 3.3.3. I don't want absolute orientation because I am operating in an area with a lots of metal and high power electrical circuits. With the magnetometer feeding into the estimation this will basically be useless to me. Maybe you can expose a way to select among the 5 fusion methods?

yoctopuce, 2017-05-03:
That could make sense, indeed, I will look at it.

boilerbots, 2017-06-12:
I noticed this request was closed, Is it now possible to modify the fusion algorithm and disable the manetometer as noted in the Bosch documentation? Maybe I should great a new feature request just for the Fusion modes?

yoctopuce, 2017-06-12:
I've closed this issue because we added this feature request on our internal todo list. But If you prefer, we can keep this issue open until we will implement this feature.

boilerbots, 2017-06-12:
You can close it since the subject does not match. I will watch the releases for the change. 

Curt

closed 2017-06-27 by yoctopuce