yoctopuce/yoctolib_cpp #25: Yocto-MaxiThermistor capturing bogus measurements

boilerbots, 2017-06-27:
I have a dozen of your Yocto-MaxiThermistor boards that are connected to computers that record the measurements at 1Hz continuously. After leaving these running for months and now analyzing the logs I find bad temperature measurements.

First I thought maybe this is a glitch in the software but when I look at the VirtualHub page it also shows the Max Temperature updating when then bad measurements occur, so it isn't my software.

Next I thought maybe it is the particular computer setup, so I connected one various boards to various other computers and after monitoring them for just 10 minutes I see the bad measurements.

So my next guess is that the firmware has a timing issue. The bad measurement seems to show up on any and all channels randomly. The attached image shows two channels that have already captured bogus measurements.

[localhost_4445_webapp.pdf](https://github.com/yoctopuce/yoctolib_cpp/files/1103792/localhost_4445_webapp.pdf)

yoctopuce, 2017-06-27:
This issue is not related to the C++ lib. It's better to continue this
discussion by email. Could you contact support@yoctopuce.com ?

boilerbots, 2017-06-27:
Will do.

closed 2017-06-27 by boilerbots