yoctopuce/yoctolib_cpp #28: Where is `ui_inventory.h`?

nadous, 2018-07-25:
In  yoctolib_cpp/Examples/Prog-QT/QTDemo/inventory.cpp, you're including an header called `ui_inventory.h`but I can't find it... Am i missing something?

octet8, 2018-07-25:
This file is generated by the User Interface Compiler (uic)  during project compilation. 

To be honest, it's the only QT project I've ever write, and I've adapted the notepad example of the QT documentation.  See: [Getting Started Programming with Qt Widgets](http://doc.qt.io/qt-5/qtwidgets-tutorials-notepad-example.html)

I've just check and this project is compiling and working on Windows using Qt 5.3.1.
```
	Qt Creator 3.1.2 (opensource)
	Based on Qt 5.3.1 (MSVC 2010, 32 bit)
```

yoctopuce, 2018-07-25:
This file is generated by the User Interface Compiler (uic)  during project compilation. 

To be honest, it's the only QT project I've ever write, and I've adapted the notepad example of the QT documentation.  See: [Getting Started Programming with Qt Widgets](http://doc.qt.io/qt-5/qtwidgets-tutorials-notepad-example.html)

I've just check and this project is compiling and working on Windows using Qt 5.3.1.
```
	Qt Creator 3.1.2 (opensource)
	Based on Qt 5.3.1 (MSVC 2010, 32 bit)
```

closed 2018-07-25 by yoctopuce

nadous, 2018-07-25:
Thanks for the quick reply. I myself is a newbie with `Qt` but you're right, somehow my `qt Creator` was missing the `uic` building step that is said to be triggered alongside `qmake`. Since I've messed with the basic qt project's directory structure, I've added a step in the toolchain, as simply as triggering: `uic -o src/ui_inventory.h inventory.ui`. Problem solved.