yoctopuce/yoctolib_cpp #30: I can't find a way to let the compass to read faster

jason19990305, 2019-06-17:
I just looked in the C++ API website but I still can't find a way to let the compass to read faster, is there any way that I can adjust the speed of the compass sensor to read data faster. thanks a lot!

yoctopuce, 2019-06-17:
You need to use the set_bandwidth() method to increase the frequency. For the compass you can increase the frequency up to 20 Hz with set_bandwidth(20). You can find more informations on our article [Tips for using the Yocto-3D-V2](https://www.yoctopuce.com/EN/article/tips-for-using-the-yocto-3d-v2)

Also if you want to be as reactive as possible do not forget to work with value callback instead of using the get_currentValue() method. See our aritcle [Pooling vs Callback](https://www.yoctopuce.com/EN/article/polling-vs-callback) that explain how to use callbacks:

jason19990305, 2019-06-19:
I use the callback. But that speed is still not faster. I use Yocto-3D.
[image]
[image]

yoctopuce, 2019-06-19:
Sadly only the Yocto-3D-V2 support the change of frequency.

Regarding your code, for maximum efficiently you should not use 
any get_xx() method in your callback. All get_xxx() method can
potentially do IO access and slow down the code.

The more efficient way to retrieve the value is to convert the
string ```value``` that you receive in parameter. 

Like so:
```cpp
static void valueChangeCallBack(YCompass *sen, const string& value)
{
    int int_val = atoi(value.c_str());
    printf("compass = %d\n", int_val);
}
```

jason19990305, 2019-06-19:
[image]

Can i use this CallBack in Gyro and get z Value ?

yoctopuce, 2019-06-19:
If you want to retrieve the orientation of the Yocto-3D, you can use the 
``registerAnglesCallback()`` method of the YGyro class. The callback function
will be called with the tree decoded angles:

```cpp

void gyroCallback(YGyro *obj, double roll, double pitch, double heading)
{
    printf("roll=%f pitch=%f heading=%f\n",
           roll, pitch, heading);
}

...

YGyro* gyro = YGyro::FirstGyro();
if (gyro) {
    gyro->registerAnglesCallback(gyroCallback);
}

```

closed 2021-05-07 by yoctopuce

yc541, 2022-01-21:
@yoctopuce 

I am having a similar problem while getting the gyro readings. I am using the exact gyroCallback above in a while loop to print the 3 angles. However it seems that the print rate can't go faster than roughly 5Hz. And there is a pretty obvious visual delay (about half a second) between the movement of the sensor and the changes of the printed values. I have tried to set the bandwidth from 20 to 100 but this doesn't seem to change the print rate. 
The gyro report rate is set to 100/s in virtual hub, and the sensor readings in virtual hub are really fast and responsive. 
[image]

yoctopuce, 2022-01-21:
@yc541: 
1) the call to registerAnglesCallback must be made only once to setup the callback, not in the loop.
2) Within the loop, remove your usleep and restore the call to YAPI::Sleep, which will listen for incoming events and trigger the angle callbacks. Without the call to YAPI::Sleep (or YAPI::HandleEvents if you prefer). you will not get your callbacks.

yc541, 2022-01-21:
Now it works perfectly, many thanks!