yoctopuce/yoctolib_cpp #31: yFindColorLedCluster() and nextColorLedCluster() throw exception on YRGBLED2 with short serial numbers

boilerbots, 2020-05-12:
We have some YRGBLED2 boards that have 5 character long serial numbers, for example YRGBLED2-C3E16. These boards can not be found using yFindColorLedCluster() or yFirstColorLedCluster() because an exception is thrown. We have lots of boards that look like YRGBLED2-121916 and they all work fine.

I just updated to the latest library release, latest VirtualHub and this is still happening, previously my library was from 2017.

boilerbots, 2020-05-12:
The exception message is:

```
terminate called after throwing an instance of 'YAPI_Exception'
  what():  Failed to parse HTTP header
Aborted (core dumped)

```

boilerbots, 2020-05-12:
Also I see in the VirtualHub output some messages about this device

```
May 12 11:04:20 r2dc071 VirtualHub[854]: [2020-05-12 10:53:12] (01) 4:4:request read for dev YRGBLED2-C3E16 error "Missing Packet (ystream:1521)" (2:57A)
May 12 11:04:20 r2dc071 VirtualHub[854]: [2020-05-12 10:54:59] (01) [1]ystream: 1519: PKTs: 3pkts (15 in / 12 out)
May 12 11:04:20 r2dc071 VirtualHub[854]: [2020-05-12 10:54:59] (01) [1]ystream: 1519: PKTs: start 100010e0 stop =10001180
May 12 11:04:20 r2dc071 VirtualHub[854]: [2020-05-12 10:54:59] (01) [1]ystream: 1519: PKTs: invalid pkt 0 (no=5 should be 3
May 12 11:04:20 r2dc071 VirtualHub[854]: [2020-05-12 10:54:59] (01) [1]ystream: 1519: PKTs: invalid pkt 1 (no=6 should be 4
May 12 11:04:20 r2dc071 VirtualHub[854]: [2020-05-12 10:54:59] (01) [1]ystream: 1519: PKTs: invalid pkt 2 (no=7 should be 5

```

boilerbots, 2020-05-13:
Digging deeper I see something strange with the various iterators.

This seems to work at first:
```
  YModule *modItr;
  modItr = yFirstModule();
  while (modItr != NULL) {
    //ROS_INFO_STREAM("Found Module: " << modItr->describe());
    string hwId = modItr->get_hardwareId();
    ROS_INFO_STREAM("HardwareId=" << hwId.substr(0, hwId.find(".module"))
          << " firmwareRelease=" << modItr->get_firmwareRelease());
    modItr = modItr->nextModule();
  }
```

Where this fails:

```
 YSensor *itr = yFirstSensor();
  while (itr != NULL) {
    string id_str = itr->get_hardwareId();
    ROS_INFO_STREAM("id_str=" << id_str);
    itr = itr->next();
  }
```
Then The first iterator loop will also fail and I have to power cycle that RGBLED board to get it to work again.

yoctopuce, 2020-05-14:
That issue is currently handled by mail. Most probably a hardware-specific issue related to hub chipset and/or host firmware/kernel causing USB packet loss

closed 2021-05-07 by yoctopuce