yoctopuce/yoctolib_cpp #35: No install targets in makefile

brian-finisher, 2021-12-23:
It would be really nice if there were standard `install` targets in the Makefile (or better yet, if you provided a CMakeLists) for yoctolib and yapi. Is there a reason this isn't the case?

In our use case, I'm building the yocto binaries in a Docker container and would really like to be able to run "make install" to copy:

- the headers to '/usr/local/include'
- executables to '/usr/local/bin'
- .so and .a files to '/usr/local/lib'
- other artifacts maybe to '/usr/local/share'

Obviously these targets could vary based on linux distrobution/OS, but the target directories should always be available as variables.

Having the files installed in the OS path as above would greatly simplify my CMakeLists, and be consistent  with other packages manually compiled and installed in our Docker containers.

srinsoz, 2021-12-24:
That's interesting. We are more used to install manually the library and headers that we need. That's why we usually do not add installation scripts in our projects.

I will try to add this feature in the in the CMakeLists. But do not expect this feature before mid January.

srinsoz, 2022-01-03:
I've just updated the CMakefLists files to support installation of the library.

To compile and install the shared version of the C++ library you just need to execute the following code
in the root of the project.
````
mkdir build
cd build
cmake ../Sources/ -DBUILD_SHARED_LIBS=ON
cmake --build .
cmake --install .
````

Does it solve your issue?

brian-finisher, 2022-01-03:
Thanks so much for your quick response and implementation! I think this should work great!
I'm building a new Docker image now to try it out and verify that the header/lib installs work. This should really clean up our CMakeLists.

brian-finisher, 2022-01-04:
One note - Need to actually run:
`make install` 
to actually copy the files to the install location.

I think in our case we'll actually prefer static libs. I could imagine that it might be nice to be able to install both shared *and* static libs. Perhaps add a separate add_library(), and append a 'static' suffix to make it clear? Not sure how this would impact other platforms, such as Windows.

brian-finisher, 2022-01-05:
It appears I now have compile errors because of missing the "yapi/yversion.h" and "yapi/yjson.h" header files.
Manually copying those seemed to resolve the problem, but should be fixed on your end.
I'm guessing in Sources/yapi/CMakeLists.txt:
`install(FILES yapi.h ydef.h yversion.h yjson.h DESTINATION include/yapi) `
And/or in Sources/CMakeLists.txt:
`install(FILES yapi/yapi.h yapi/ydef.h yapi/yversion.h yapi/yjson.h DESTINATION include/yapi)`

I'm also getting a bunch of undefined references by the linker (using the static .a libs), but suspect that is on my end. I'll try tinkering with my CMake files to get those resolved tomorrow.
Thanks again!

srinsoz, 2022-01-05:
By default cmake will use and install the static version of the library. I thought you wanted to use shared library so I added the ``-DBUILD_SHARED_LIBS=ON`` option in the example. But if you want to use the static lib (which I recommend), just use 
``cmake ../Sources/`` or ``cmake ../Sources/ -DBUILD_SHARED_LIBS=OFF``.

Regarding the missing header, I have also find this issue and I have push the fix.

brian-finisher, 2022-01-13:
Sorry for the long delay - finally got back to this today and got everything to compile against the installed static headers! Thanks so much!

closed 2022-01-14 by srinsoz