yoctopuce/yoctolib_cpp #36: CMAKE install direct

peppedxAlto, 2023-06-30:
Why are you disallowing me to choose my installation prefix?

if(${CMAKE_SYSTEM_NAME} STREQUAL "Linux")
SET(CMAKE_INSTALL_PREFIX "/usr")
endif()

srinsoz, 2023-07-03:
We have set this variable to support debian based distribution. But I understand that
can be annoying to prevent the user to specify the destination. Does remove these tree
lines solve your issue?

peppedxAlto, 2023-07-03:
Hi,
Yes it does, but honestly I think this goes against the base expectation of cmake users.

Also I'd suggest for the cmake file to live in the root directory and add the installation targets so as to ease consuming the package using Cmake FetchContent extension.

Anyway thank you for the support.

peppedxAlto, 2023-07-03:
I haven't been entirely correct, installation targets are there but ".cmake" files could be generated ( https://cmake.org/cmake/help/latest/manual/cmake-packages.7.html#creating-packages ).

srinsoz, 2023-07-04:
I think the following would work for everybody.
```
if(CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT AND ${CMAKE_SYSTEM_NAME} STREQUAL "Linux")
SET(CMAKE_INSTALL_PREFIX "/usr")
endif()
```
See more information  on CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT  : https://cmake.org/cmake/help/latest/variable/CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT.html

peppedxAlto, 2023-07-04:
Yep,
for sure it works for me.

srinsoz, 2023-08-03:
We have just released the fix.

closed 2023-08-03 by srinsoz